chilipeppr/robot-actuator-esp32-v8 #1: Small stepper motor torque rating?

Donald-Nordwick, 2019-10-24:
Do you know what the relative torque is for the smaller stepper motors from aliexpress? https://www.aliexpress.com/item/2054451318.html

I bought a few stepper motors from stepperonline that are rated at 7 Ncm and they're having issues even after using some grease.
https://www.omc-stepperonline.com/il-giro-nema-14-bipolare-0-9deg-5ncm-7-08oz-in-0-5a-8-5v-f36x12mm-4-fili.html?search=Round%20Nema%2014%20Bipolar%200.9deg%207Ncm%20%289.91oz.in%29%200.5A%208.5V%20%CE%A636x12mm%204%20Wires

chilipeppr, 2019-10-24:
Look at that moon's stepper motor I put in the bom.  The ones I got from AliExpress have very similar characteristics. The omc one you're showing is 0.9 degree instead of 7.8 degree so that is a big difference and mine are all about 150ma and yours is 500ma which makes me think yours is way stronger.

Did you break in your gears? I find that I sometimes have to print the planetary gears at 99 percent depending on printer differences.

chilipeppr, 2019-10-24:
I am not able to stop the bare shaft with my fingers on my small steppers.

Donald-Nordwick, 2019-11-05:
Thanks for the response. I was able to get it to move, I just have to work on getting those gears filed down. I was looking at step pulse loopback on eagle for the pcb and its listed as 0 ohm. Is that a typo?

chilipeppr, 2019-11-05:
The step pulse loopback is not a typo. It is a loopback. The track overlaps to connect the nets it if you cut the track out a 0 ohm to reconnect