chilipeppr/robot-actuator-esp32-v8 #3: Error with rmttx_stepper_queue_v4

mcmadhatter, 2019-11-26:
I tried uploading and running the lua code, but I get and error on startup

[0;32mI (468406) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
Slept DRV8825
loop or previous error loading module 'rmttx_stepper_queue_v4'

chilipeppr, 2019-11-26:
You have to upload all of the code in the lua folder to your ESP32 first,
make sure it's compiled to .lc files, then you won't get that error because
it will find the library.

mcmadhatter, 2019-11-26:
I've uploaded the 13 lua files that are in the repo, and compiled all to .lc flies. 

The only ones that are in the chilipeppr.png that are not in the repo are fan_temp_ctrl.lua tmp36_v2.lua gcode_jog_btns.lua and gcode.txt
[image]

chilipeppr, 2019-11-27:
Ok, just re-checked in my latest code. What a pain transferring out of ChiliPeppr into Github. I need a better way to use a Github repo in ChiliPeppr!

This is a pretty big difference using my latest code. The main entry point is main_cayenn_robot_v5.lua as you'll see in init.lua. This is now using Cayenn so you can see the actuator show up in the Cayenn widget in ChiliPeppr inside the ESP32 workspace and the robot workspace, which is working kinda/sorta now with the Axes widget in upper right corner.

Let me know how it goes.

[image]

mcmadhatter, 2019-11-27:
Hi , with the latest it gets a bit further before a kernel panic. I should be using the NodeMCU-touch-pulsecnt-rmttx.bin in the firmware folder shouldn't I?

NodeMCU ESP32 build unspecified powered by Lua 5.1.4 on SDK IDF
[0;32mI (336) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
Slept DRV8825
Initted fan & temp sensor. Freq:30, Duty:8192
Apppending
Initted Gcode file save/retrieve
Initting ws2812. Using channel 7. GPIO 26. memBlocks 1
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40147b01  PS      : 0x00060a30  A0      : 0x80147cd0  A1      : 0x3ffbc1b0  
A2      : 0x3ffc0d3c  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3ffcd944  
A6      : 0x3ffcd954  A7      : 0x00000000  A8      : 0x80147491  A9      : 0x3ffbc160  
A10     : 0x3ffc7f58  A11     : 0x3ffc15c8  A12     : 0x00001800  A13     : 0x00000800  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd  

ELF file SHA256: a239507fd8d32e86f79312bdb2015e39d7edf86abcf3757da8f7e68d121d0aac

Backtrace: 0x40147b01:0x3ffbc1b0 0x40147ccd:0x3ffbc1d0 0x401472f7:0x3ffbc200 0x40145470:0x3ffbc220 0x400d49c9:0x3ffbc240 0x4013d777:0x3ffbc270 0x40143b61:0x3ffbc2b0 0x4013d84e:0x3ffbc300 0x401452a5:0x3ffbc320 0x40168afe:0x3ffbc340 0x4013d777:0x3ffbc360 0x40143b61:0x3ffbc3a0 0x4013d84e:0x3ffbc3f0 0x401447cd:0x3ffbc410 0x4013d095:0x3ffbc430 0x4013d936:0x3ffbc4b0 0x401452e6:0x3ffbc4e0 0x4013f57c:0x3ffbc510 0x4013f86d:0x3ffbc530 0x4013d777:0x3ffbc560 0x4013d845:0x3ffbc5a0 0x40144811:0x3ffbc5c0 0x4013d095:0x3ffbc5e0 0x4013d936:0x3ffbc660 0x40145316:0x3ffbc690 0x4013f9f6:0x3ffbc6c0 0x400f8f51:0x3ffbc6f0 0x400d269f:0x3ffbc730

Rebooting...

chilipeppr, 2019-11-27:
Hmm. I better send you my latest firmware. It could be that.

chilipeppr, 2019-11-27:
ok, just checked in my oct 4th firmware build. try that.

mcmadhatter, 2019-11-27:
Hi, I flashed the new firmware

```
 ~/Programming/robot-actuator-esp32-v8/firmware   master ● ⍟2  esptool.py --port /dev/cu.Repleo-CP2102-01DB132C write_flash 0x1000 bootloader.bin 0x10000 NodeMCU-touch-pulsecnt-rmttx.bin 0x8000 partitions_singleapp.bin
esptool.py v2.6
Serial port /dev/cu.Repleo-CP2102-01DB132C
Connecting.....
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 24:0a:c4:c6:7b:f0
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 22656 bytes to 14177...
Wrote 22656 bytes (14177 compressed) at 0x00001000 in 1.2 seconds (effective 145.1 kbit/s)...
Hash of data verified.
Compressed 998560 bytes to 639050...
Wrote 998560 bytes (639050 compressed) at 0x00010000 in 56.6 seconds (effective 141.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 1668.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
interestingly it now crashes sooner than with the old firmware:

`Mounting flash filesystem...
I (418) wifi: wifi driver task: 3ffc321c, prio:23, stack:3584, core=0
[0;32mI (418) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (418) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
I (438) wifi: wifi firmware version: ec61a20
I (438) wifi: config NVS flash: enabled
I (438) wifi: config nano formating: disabled
I (448) wifi: Init dynamic tx buffer num: 32
I (448) wifi: Init data frame dynamic rx buffer num: 32
I (448) wifi: Init management frame dynamic rx buffer num: 32
I (458) wifi: Init management short buffer num: 32
I (458) wifi: Init static rx buffer size: 1600
I (468) wifi: Init static rx buffer num: 10
I (468) wifi: Init dynamic rx buffer num: 32

NodeMCU ESP32 build unspecified powered by Lua 5.1.4 on SDK IDF
[0;32mI (498) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
Slept DRV8825
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
`

mcmadhatter, 2019-11-27:
Just a thought can esptool dump the filesystem from an esp32, so it can be flashed to another esp32?

chilipeppr, 2019-11-27:
I'm not sure if esptool can do that, but there is an Upload All button in
ChiliPeppr to send all of that code to the ESP32 without much effort other
than just waiting around for it all to upload. It makes a nice ding sound
when it's done uploading all of the files.

chilipeppr, 2019-11-27:
[image: image.png]

chilipeppr, 2019-11-27:
On your error, it does look like you're seeing a  SW_CPU_RESET which is
weird to me. That's as if it ran code that did a node.restart()

mcmadhatter, 2019-11-27:
esptool.py --port <your port>  read_flash 0 0x400000 flash_contents.bin   will dump everything including the filesystem.

Would you be ok to try doing that on your working chip, and sending me the flash_contents.bin to try on mine?

chilipeppr, 2019-11-28:
> esptool.py --port read_flash 0 0x400000 flash_contents.bin will dump

I'm not near any of my ESP32 devices, sorry.

mcmadhatter, 2019-11-28:
OK, no worries.